roleoroleo/yi-hack-MStar #545: SAVE CONFIGURATION issue

darius456, 2024-05-28:
Hi all,

Firmware Version 0.5.5
Base Version 4.6.0.0A_201908271549
Model Suffix h201c

When i open ONVIF configuration page i cant save settings.

In web browser i got this:
_module configurations_onvif not registered
ruleApplier.js:1 Uncaught SyntaxError: Identifier 'observeDomChanges' has already been declared (at ruleApplier.js:1:1)
VM258:1 Uncaught TypeError: Cannot redefine property: googletag
    at Function.defineProperty (<anonymous>)
    at <anonymous>:1:196
    at <anonymous>:1:524
(anonymous) @ VM258:1
(anonymous) @ VM258:1_

Save button work ok on "Configurations" page.

roleoroleo, 2024-05-28:
Please, try to clean the cache of your browser.
I changed js files.

darius456, 2024-05-28:
> Please, try to clean the cache of your browser.
This fix the problem.

closed 2024-05-28 by darius456