roleoroleo/yi-hack-MStar #552: can't flash BFUS 4.5.0.0

Meehou, 2024-10-25:
hello

I have camera BFUSY21PX4 with firmware 4.5.0.0C_201910080934 and I can't flash it, coppied 2 files y203c on formated card in fat32, insert to camera nad nothing happened, it's starting in few seconds like normal without card

any help?

roleoroleo, 2024-10-25:
It should work.
Try another sd card.

vil-lian, 2025-02-21:
I just came across this issue as well. Tried different sd card, factory reset of the camera, nothing seems to get them looking for those firmware files. Just boots up normally.

vil-lian, 2025-02-25:
I'm stupid. The flashing happened so quick that I assumed it *wasn't* flashing the camera. Tried the web gui and, lo and behold, yi-hack appears.  Thanks for replying roleoroleo, appreciate the work.