roleoroleo/yi-hack-MStar #556: reboot loop h201c (dome1080p / bfus)

thertp, 2024-11-08:
After updating from 0.4.7 to 0.5.5, I'm getting a reboot loop.

Finally, found time to look at that. For now, in ``/tmp/log.txt``, the last message is "rcd may crashed".
Digging a little bit more, I guess it's ``rcd`` is ``mp4record``.
I can see it in ``ps`` at boot, so either it's crashing later or there's communication issue between ``dispatch`` and ``mp4record``
(over the ``ipc_rcd`` file?).
Not sure what would be the next step. Any hint of what to do next ?

_While it may be a hardware bug, I find it unlikely as the issue appeared after the 0.5.5 upgrade._

roleoroleo, 2024-11-11:
The message "rcd may crashed" happens very frequently.
It shouldn't be a problem.
The bootloop could be a problem related with the free space in the root partition.
Can you try to restore 0.4.7 and the upgrade to 0.5.5 step by step?
I mean 0.4.8, 0.4.9 and so on...

thertp, 2024-11-13:
ok. I'll try to find time to do that. Can I do something clever like bisecting the image versions instead of upgrading with each image ?

roleoroleo, 2024-11-15:
There are 2 different tests:
1 - You could try to downgrade to a previous version and then upgrade again to 0.5.5, to test  if it's a free space problem.
2 - If it doesn't work, I need to know from which version starts the problem. In this case you should upgrade step by step.

thertp, 2024-11-26:
I had few mins available yesterday. I found out that:
- alas, the disk space is not the issue (connected through ssh and run df),
- due to lack of time, I didn't try flashing other versions but looks like some old issues similar are found even at 0.4.8
- after replacing /sbin/reboot with a shell script, looks like it's ``rmm`` which is triggering the reboot. As my script was not rebooting the camera, it did run overnight **but** I was not able to get a rtsp stream.

roleoroleo, 2024-11-27:
There is a watchdog script (wdrtsp.sh) that monitors rmm and reboot the cam if rmm dies.
If rmm is not running you can't have the stream.
If this is the case, the question is why rmm dies, Usually it happens when the device is running in low memory.
Try to disable all unneded processes.

thertp, 2024-11-28:
iirc, I've already disabled the wdrtsp (one of the first thing I did when getting the reboot loop). The rmm process is not dying, but it's doesn't seem to work either...

thertp, 2024-11-28:
btw, I'm seeing in the logs a bunch of "invalid msg" from dispatch. Is there some doc about all that stuff ?

roleoroleo, 2024-11-30:
> btw, I'm seeing in the logs a bunch of "invalid msg" from dispatch. Is there some doc about all that stuff ?

No log sorry...
Are you able to restore the stock fw?

woniuzfb, 2024-12-30:
Does anyone has a mtd4 partition backup? I have the same problem

roleoroleo, 2024-12-30:
Probably you have a different cam.
mtd4 is the vdl partition (where Yi stores cam settings).
rootfs is mtd2 and home fs is mtd3.

woniuzfb, 2024-12-30:
> Probably you have a different cam. mtd4 is the vdl partition (where Yi stores cam settings). rootfs is mtd2 and home fs is mtd3.

I knew that, but it must have been corrupted, which is why the camera kept rebooting in a loop.

thertp, 2024-12-31:
I fear this bug will have to be closed. I tried last week to boot again my cam but it wouldn't start. I've connected cables to the serial port and nothing is showing. I guess the cam is dead :(

roleoroleo, 2025-01-01:
> I knew that, but it must have been corrupted, which is why the camera kept rebooting in a loop.

mtd4 is different for each cam.
You should try to restore a backup through the serial interface.

guilhem, 2025-02-26:
I just flashed 0.5.5 from stock and I have the reboot loop.
I tried to flash 0.4.7, same problem :/

roleoroleo, 2025-02-28:
Use the unbrick procedure described in the wiki.

closed 2025-06-29 by github-actions[bot]