roleoroleo/yi-hack-MStar #559: Status page 404 with v0.5.5

olicooper, 2024-12-31:
Index / status page does not load in firmware version 0.5.5 but worked okay in version 0.4.9
Port change from 80 back to original port 8080 doesn't make a difference.

Console logs are as follows

v0.5.5
```log
Navigated to http://192.168.43.54/index.html
Current page: status
GET http://192.168.43.54/pages/status.html?nocache=1735648497 404 (Not Found)
```

v0.4.9
```log
Navigated to http://192.168.43.54:8080/index.html
Current page: status
```

roleoroleo, 2025-01-01:
Is the cam working properly (apart from web interface)?

olicooper, 2025-01-01:
Not really, but I'm not sure if the issues are related or if the camera is failing because it has been in continuous use for 4+ years and the flash is worn out? It is deployed in another location which makes testing difficult. Initially it appeared to be okay for 24 hours after upgrading but it failed to reboot after plugging in a network cable the next day to make it more reliable.

There were no boot sounds, only a solid red indicator light. I tried diagnosing the problem by swapping the power supply etc. but after a few power cycles it weirdly started working again after leaving it plugged in for 20 minutes (the firmware files were still on the SD card). The settings had gone back to default though too.

Because of the web page issue and the strange boot hanging issue I decided to restore the old firmware which has been fine ever since. I haven't tried incrementing the versions but testing is difficult to do remotely.

closed 2025-05-02 by github-actions[bot]