roleoroleo/yi-hack-MStar #562: Onvif RSTP straming

idanny76, 2025-01-28:
I have a new NVR and the Yi Cameras have great signal, but the the video feed freezes or lags. I am using the RSTP Server... not the alternative, since the alternative does not work with the NVR. The NVR fount the cameras through Onvif. I have RSTP enable as well as Onvif. Not sure if Onvif will work without the RSTP Server enabled. Not sure why the lag on the streaming. Any suggestions.... Or is it time for these old Yi PTZ 1080p cameras to be replaced?

roleoroleo, 2025-01-28:
Onvif is not responsible for this issue. It's probably related to rtsp.
But I spent a lot of time to optimize the streaming.
How many clients are you using?
Is the cam suffering in terms of resources (ram or cpu)?
Is the problem present with other rtsp clients?

idanny76, 2025-01-28:
I have a total of 7, but I went ahead and upgraded to new ones from the NVR brand. The new ones have 2.4 & 5GHz band and are 5MP picture quality. This Yi cameras are 1080p, I have had it for a very long time and its time to upgrade. Thanks anyways for your support of the Yi Hack firmware and always ready to help. Thank you.

closed 2025-05-29 by github-actions[bot]