roleoroleo/yi-hack-MStar #563: Camera no longer supplies the hostname to the dhcp server

olivluca, 2025-02-02:
I was using an old 0.4.x version and I just updated to the latest one (0.5.5).
Now the camera doesn't supply the correct hostname to the dhcp server, that's because in the script `/home/yi-hack/script/wifidhcp.sh` reads it from `/tmp/sd/yi-hack/etc/hostname` instead of from `/home/yi-hack/etc/hostname` (where the hostname is saved from the web interface).
That was done in commit 9cf96db, why?

olivluca, 2025-02-14:
Ouch, you're right, the commit is correct, the problem is that the script in the camera points to the wrong file (/tmp/sd/yi-hack/etc/hostname).
Now I see that the commit was made on may 28 while 0.5.5 was released on april 11, so the fix is not there.
The wrong script was introduced with commit f37e918

roleoroleo, 2025-02-16:
Yes, there was a copy/paste error from allwinner repo.
Use the current version of wifidhcp.sh.
It's ok.