roleoroleo/yi-hack-MStar #564: FTP not working

Pmachat, 2025-02-27:
Hi,
Although I configured the download service via FTP, it does not work.

[image]

As soon as I manually launch the Ftppush.sh script it works.

[image]

My configuration

[image]

Could you help me ?

Thanks

roleoroleo, 2025-02-28:
Check if the script is properly set in the crontab:
```
root@yi-hack-h52ga:~# cat /tmp/var/spool/cron/crontabs/root
```

Pmachat, 2025-02-28:
Hi,
I must have made a mistake. I can no longer connect in root mode (either SSH or FTP).
I reset the camera, the problem remains (without password)
An idea ?
Thanks

Pmachat, 2025-02-28:
I found the solution. I upgraded the version to 0.5.5. Everything is alright.
Thanks for the help

closed 2025-02-28 by Pmachat